Mesh4all/m4a-firmware #27: docs: Add Contibutors File

eduazocar, 2021-11-15:
#### Description

I need to start adding contributors file, please everyone, could you please send me your info, github username, email, name, any content to be used as an identifier

luisan00, 2021-11-15:
> ... github username, email, name, any content to be used as an identifier

If name, user and  email then :

| Name | Github User | email |
|:---|:---|:---|
| Luis A. Ruiz | @luisan00 | luisan00@hotmail.com |

if it's two fields then take my name and email @eduazocar

eduazocar, 2021-11-15:
Thanks @luisan00

luisan00, 2021-11-15:
Please remove fossa... It's only a bot,, but add @Feedurhead-1

luisan00, 2021-11-17:
@eduazocar  this should be driven by writing in the PR description(as described in the PR template) Fix #[issue number], for automatic resolution. if not, you need to close by hand after merging

closed 2021-11-17 by luisan00