Mesh4all/m4a-firmware #34: Make runners great again

luisan00, 2021-11-17:
#### Description
Looks like after merging the static tests, some structures are raising positive, review and fix each positive by running locally the `static-test` tool

### Useful links
- PR: test #25 
- PRs: Runner fail: between  #26 and #32, both included

eduazocar, 2021-11-22:
errors checkshell:
ine 23: /home/deskto/Documents/m4a-firmware/dist/tools/shellcheck/dist/tools/ci/changed_files.sh: No such file or directory
line 24: /home/desktop/Documents/m4a-firmware/dist/tools/shellcheck/dist/tools/ci/github_annotate.sh: No such file or directory

errors Vera++: 
error: couldn't open "boards/m4a-mb/board.c": no such file or directory
    while executing
"open $fileName "r""
    ("foreach" body line 6)
    invoked from within
"foreach fileName [getSourceFileNames] {
    if { $fileName == "-" } {
      # can't check the content from stdin
      continue
    }
    set file [op..."
    (file "/home/bucare/Documents/M4ALL/m4a-firmware/dist/tools/vera++/scripts/rules/F001.tcl" line 4)

the coccinelle test: 
--- tests/periph_uart/main.c
+++ /tmp/cocci-output-70713-ce9986-main.c
@@ -41,7 +41,7 @@
 #define MAX_SENTENCE_SIZE 128
 
 #define MAIN_QUEUE_SIZE (8)
-static msg_t _main_msg_queue[MAIN_QUEUE_SIZE];
+msg_t _main_msg_queue[MAIN_QUEUE_SIZE];
 
 size_t count = 0;
 char uart_buff[MAX_SENTENCE_SIZE] = {0};
Python path configuration:
  PYTHONHOME = '/lib/x86_64-linux-gnu/..'
  PYTHONPATH = '/usr/bin/../lib/coccinelle/python'
  program name = 'python3'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/bin/python3'
  sys.base_prefix = '/lib/x86_64-linux-gnu/..'
  sys.base_exec_prefix = '/lib/x86_64-linux-gnu/..'
  sys.executable = '/usr/bin/python3'
  sys.prefix = '/lib/x86_64-linux-gnu/..'
  sys.exec_prefix = '/lib/x86_64-linux-gnu/..'
  sys.path = [
    '/usr/bin/../lib/coccinelle/python',
    '/lib/x86_64-linux-gnu/../lib/python38.zip',
    '/lib/x86_64-linux-gnu/../lib/python3.8',
    '/lib/x86_64-linux-gnu/../lib/python3.8/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007fe4828a6180 (most recent call first):
<no Python frame>

licenses Errors:
ls: cannot access './patterns': No such file or directory
file has an unknown license header: 'boards/m4a-mb/board.c'
file has an unknown license header: 'boards/m4a-mb/include/board.h'
file has an unknown license header: 'boards/m4a-mb/include/gpio_params.h'
file has an unknown license header: 'boards/m4a-mb/include/periph_conf.h'
file has an unknown license header: 'firmware/include/main.h'
file has an unknown license header: 'firmware/main.c'
file has an unknown license header: 'tests/board_stm32f469i-disco/main.c'
file has an unknown license header: 'tests/driver_ds18/main.c'
file has an unknown license header: 'tests/driver_epd-bw-spi/epd_bw_spi_params.h'
file has an unknown license header: 'tests/driver_epd-bw-spi/external_drivers/epd_bw_spi/epd_bw_spi.c'
file has an unknown license header: 'tests/driver_epd-bw-spi/external_drivers/epd_bw_spi/include/epd_bw_spi.h'
file has an unknown license header: 'tests/driver_epd-bw-spi/external_drivers/epd_bw_spi/include/epd_bw_spi_internal.h'
file has an unknown license header: 'tests/driver_epd-bw-spi/external_drivers/epd_bw_spi/include/epd_bw_spi_params.h'
file has an unknown license header: 'tests/driver_epd-bw-spi/external_drivers/epd_bw_spi/include/epd_bw_spi_pictures.h'
file has an unknown license header: 'tests/driver_epd-bw-spi/main.c'
file has an unknown license header: 'tests/driver_u8g2/main.c'
file has an unknown license header: 'tests/periph_uart/main.c'
file has an unknown license header: 'tests/unittests/main.c'
file has an unknown license header: 'tests/unittests/map.h'

luisan00, 2021-11-23:
it seems you need to resolve some dependencies @eduazocar

PR #46 (2021-11-24): ci: Static-tests resolving the doccheck errors

### Contribution description

PR designated to resolve the errors and warnings from static-tests doccheck,  

### Testing procedure

General folder (m4a-firmware):
```sh
make static-test
```
Specific tool:
```sh
cd dist/tools/doccheck
./check.sh
```
### Issues/PRs references

Related to  #41 #40 
Fix #34
closed 2021-11-24 by luisan00