Mesh4all/m4a-firmware #127: convert literal to string error

xkevin190, 2022-02-04:
#### Description

making the modifications so that the wifi module would take the data from the nvs, I was having problems when I set a data in the nvs it replaced the others.

#### Steps to reproduce the issue

For reproduce this issue you can  do simple printf to result optained of `nvs_key_to_name` for example:

#### Expected results

the code that was failing is this:

```c

#define literal_to_str(literal) #literal                /*!< Transform literal to string */
#define nvs_key_to_name(...) literal_to_str(__VA_ARGS_) /*!< A nonsense for the nvs component */
#define at_cmd_to_name(...) literal_to_str(__VA_ARGS__) /*!< Another one for the uart(at) component*

```

luisan00, 2022-02-04:
please can you paste the error?

luisan00, 2022-02-04:
Thanks a lot for the report! 
@xkevin190 Fixed into #128 , please try again compiling the related test

PR #128 (2022-02-04): wifi-subsys: fix wrong macro literal to string
### Contribution description
This fix a bug related to converting literals to string
### Testing procedure
Type and execute the following in the /test folder:
```sh
idf.py build -D "TEST_COMPONENTS=storage" flash monitor
```
Select the test case `#1` , the expected result should be:
```sh
Running Literal to String...
I (346) TEST: stringlify: TEST ?= expected: TEST
...m4a-firmware/wifi-subsys/components/storage/test/test_nvs.c:20:Literal to String:PASS
```
### Issues/PRs references
Fix #127
closed 2022-02-04 by luisan00